zivid/zivid-ros #6: Make an initial release?

130s, 2019-10-08:
Thanks for opensourcing the ROS driver!

For production system usecase, I think making the package referrable by specific version number would be great. Would it be possible to make a release?

By "release" I'd say 2 things:
- For general `git` project, making `git tag` is minimum requirement.
- For catkin package, updating the version in `package.xml` helps the downstream projects to utilize tools available in ROS (e.g. `roslaunch` will be able to access the package version and prints it upon boot).

And lastly, it'll be much convenient if binary of the package is available. For ROS package it can be easily done without hosting buildfarm etc. This wiki describes how-to http://wiki.ros.org/ROS/ReleasingAPackage

apartridge, 2020-01-02:
Hello @130s , sorry for the very late reply to this issue.

Going forward, when we update the version number we will also make a git tag for it. We will most likely increase the version number to 1.0.0 within a few days or so.

Regarding binary packages, we agree that this would be more convenient for deployment. The reason we currently do not support this is that our Zivid SDK API is not ABI stable. This means that we will need to build and package ROS driver towards all supported Zivid SDK versions and supported operating systems each time the ROS driver or SDK is updated. To accomplish this we need an automated build and deploy system for the ROS driver, which we have not yet implemented. This is something that we will consider in the future, however no timeline on this as of now.

tfoote, 2025-02-26:
The ROS buildfarm will automatically rebuild packages when there's new releases automatically. We can make this happen assuming that the SDK can be made available appropriately. As a fallback it would be great to at least have the source and doc entries in the rosdistro to find the driver via index.ros.org https://docs.ros.org/en/rolling/How-To-Guides/Releasing/Index-Your-Packages.html 

I'd be happy to talk directly more to help find a solution for this as it will make things much easier for users.

gavanderhoorn, 2026-11-16:
Having just `zivid_interfaces` on the buildfarm would already be massively helpful.

Tracking in #172.