zivid/zivid-ros #16: No URDF file provided for TF ros

tyseng92, 2020-02-21:
I have tried to integrate Zivid camera into my robotic system, replacing the Zed camera. I found out that this package does not contain urdf file for Zivid. Could you provide urdf file of Zivid and import it into rviz so that I can work on TF transformation in ROS? Hopefully the results will be something similar to this [zed ros package with urdf](https://github.com/stereolabs/zed-ros-wrapper/tree/master/zed_wrapper/urdf).

runenordmo, 2020-03-04:
Hi, @tyseng92. We'll consider adding such a `urdf` file, it could be useful for working with TF transforms, as well as rviz visualization / Gazebo simulation. We can provide necessary information if you are writing your own at the moment.

Could you elaborate on which fields you would like to see in the `urdf` file? What is the specific transform you need? And are you also using `urdf` for things like meshes for visualization or geometry for collision boxes?

_- Rune Nordmo, Software Engineer at Zivid_

dave992, 2020-03-05:
I recently made a description of the Zivid camera (Visual + Collision) for our own project that integrates the Zivid on a robot using ROS. The description includes the projector and camera frames, visual geometry and simplified collision geometry. 

I am now checking the package and pushing it to our fork. I hope to create a pull request with the added description package in a bit.