zivid/zivid-ros #20: Wrong dependency packaging

jonazpiazu, 2020-05-27:
This issue is not strictly related to the ROS driver, so maybe it is not the best place to be reported.

The [Toshiba Teli driver package](http://www.zivid.com/hubfs/softwarefiles/releases/1.8.1+6967bc1b-1/u16/zivid-telicam-driver_3.0.1.1-1_amd64.deb?hsLang=en) available for download from the Zivid's download page contains the file `/root/.wget-hsts` which should not be there. It is not related to the driver, and is a file generated used by `wget`, so it does not make sense to pack it. 

Moreover, it might overwrite that file if it is already existing, or cause conflicts in installation.

I have only tested the version for Ubuntu 16.

nedrebo, 2020-05-27:
Thank you for reporting this. We have this in our internal bug tracker: I will link our issue to this issue.

jonazpiazu, 2020-06-04:
Any update on this issue, or an estimate of when it can be fixed?

knatten, 2020-06-05:
Hi @jonazpiazu, I'm looking into this now. I expect to have it fixed in the official Zivid 2.0.0 release. (It's not yet fixed in 2.0.0-beta-1 which was released this week).

jonazpiazu, 2020-06-08:
So then I can assume that the fix will not be applied to version 1.8.x of the SDK?

knatten, 2020-06-08:
This issue has now been fixed, and will be part of the official 2.0.0 release. It will unfortunately not be fixed for 1.8.x.

closed 2020-06-08 by knatten