zivid/zivid-ros #33: Command/sample taking applying settings from YML file

runenordmo, 2020-08-17:


nedrebo, 2020-08-17:
Can you be more verbose? I am not sure what this is about.

runenordmo, 2020-08-17:
I think it's  quite involved providing settings for HDR in the ros sample.

We can add something in line with [the cpp sample capturing with samples from YML](https://github.com/zivid/zivid-cpp-samples/blob/master/source/Camera/Basic/CaptureWithSettingsFromYML/CaptureWithSettingsFromYML.cpp), or a command that let's you capture with settings from YML in ros.

Will probably require that #31 has been done first. This was part of a discussion in https://github.com/zivid/zivid-ros/pull/32#discussion_r469183151.

apartridge, 2020-08-17:
I agree that this should be improved. In SDK 2 we contain the whole HDR setting in one file, so we could add a command to load settings from a file, maybe also a launch parameter to initialize the settings.

closed 2021-11-10 by apartridge