zivid/zivid-ros #37: Compilation errors

smits, 2020-10-30:
When compiling against ROS kinetic in Ubuntu 16.04 using c++ (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609
I get the following error:
```
Errors     << zivid_camera:make /home/intermodalics/zivid-ros-workspace/logs/zivid_camera/build.make.000.log
/home/intermodalics/zivid-ros-workspace/src/zivid-ros/zivid_camera/src/settings_generator.cpp: In function ‘std::__cxx11::string {anonymous}::zividSettingsTypeName()’:
/home/intermodalics/zivid-ros-workspace/src/zivid-ros/zivid_camera/src/settings_generator.cpp:106:6: error: expected ‘(’ before ‘constexpr’
   if constexpr (std::is_same_v<SettingsRootGroup, Zivid::Settings> ||
      ^
```

nedrebo, 2020-10-30:
You must use a more recent compiler. Instructions are listed here: https://github.com/zivid/zivid-ros#c-compiler

smits, 2020-10-30:
I missed that, sorry about the noise.

closed 2020-10-30 by smits

nedrebo, 2020-10-30:
Np, good luck with testing it out.