zivid/zivid-ros #41: "acquisition_0_client.setConfiguration" failure in the CI

apartridge, 2021-04-14:
We quite frequently get this in the CI now. Not always, but regularly, often in different tests but same failure type.

https://dev.azure.com/zivid-devops/zivid-ros/_build/results?buildId=6069&view=logs&jobId=1a92118a-31b6-59d3-6f46-005b1de66b34&j=1a92118a-31b6-59d3-6f46-005b1de66b34&t=4902d933-60c3-56fe-9a17-c7813a975418

```
[zivid_camera.rosunit-zivid_camera_test/testCapture3DColorImage][FAILURE]-------
/root/catkin_ws/src/host/zivid_camera/test/test_zivid_camera.cpp:131
Value of: acquisition_0_client.setConfiguration(acquisition_0_cfg)
  Actual: false
Expected: true
```

apartridge, 2021-06-23:
This issue seems to not happen anymore in the CI, maybe something changed upstream. It has worked in the last 24 or so builds, according to https://dev.azure.com/zivid-devops/zivid-ros/_build?definitionId=1&_a=summary&repositoryFilter=1&view=runs. So I am closing this issue.

apartridge, 2021-09-01:
This has started happening again after the move to Github Actions.

torbsorb, 2022-04-22:
For the capture service we add a short_wait_duration.sleep()afterwards. In
```cpp
    zivid_camera::SettingsAcquisitionConfig acquisition_0_cfg;
    ASSERT_TRUE(acquisition_0_client.getDefaultConfiguration(acquisition_0_cfg, dr_get_max_wait_duration));
    acquisition_0_cfg.enabled = true;
    ASSERT_TRUE(acquisition_0_client.setConfiguration(acquisition_0_cfg));
```
Could it be that `dr_get_max_wait_duration` is a little too short? Or does it fail if that expires? Is this timeout needed for `setConfiguration` too maybe?

apartridge, 2024-08-05:
This has not happened for a while, and we have updated master to ROS2 now. Closing this issue

closed 2024-08-05 by apartridge