zivid/zivid-ros #49: Slightly misleading CaptureAssistantSuggestSettings service definition

danielcranston, 2021-08-04:
Hi, thanks for making this repo!

I noticed the [CaptureAssistantSuggestSettings service definition](http://docs.ros.org/en/melodic/api/zivid_camera/html/srv/CaptureAssistantSuggestSettings.html) mentions

```
duration max_capture_time
```

The type (`duration`) here I assumed refers to http://docs.ros.org/en/melodic/api/std_msgs/html/msg/Duration.html, but that does not seem to be the case. Consider this minimal example: 

```
import rospy
from zivid_camera.srv import CaptureAssistantSuggestSettings, CaptureAssistantSuggestSettingsRequest
from std_msgs.msg import Duration

service_proxy = rospy.ServiceProxy("/zivid/capture_assistant/suggest_settings", CaptureAssistantSuggestSettings)

request = CaptureAssistantSuggestSettingsRequest(max_capture_time=Duration(2)) <---- 
service_proxy.call(request)
```

The service call on the last line fails with `AttributeError: 'Duration' object has no attribute 'secs'`. If you change the request however:

```diff
+ request = CaptureAssistantSuggestSettingsRequest(max_capture_time=Duration(2))
- request = CaptureAssistantSuggestSettingsRequest(max_capture_time=rospy.Duration(2))
```

then the service call succeeds. I expect however that this is just duck typing at work and that this is not the intended way to create the request (`rospy.Duration` is not a ROS message type).

So my question is: what ROS message is `duration` referring to in the service definition?

apartridge, 2021-08-09:
Hi @danielcranston .

From http://wiki.ros.org/ROS/Tutorials/CreatingMsgAndSrv one can see that `duration` is a supported basic type, like ints and string. It is created in Python using rospy.Duration and in C++ using ros::Duration. Also see https://roboticsbackend.com/ros-duration-roscpy-roscpp/.

As you can see from http://docs.ros.org/en/melodic/api/std_msgs/html/msg/Duration.html, it is a message type that contains one of these basic durations as its only member. We are not using that message in our repo now.

danielcranston, 2021-08-10:
Thanks for the reply! Yep this turned out to be a pure ROS question, and indeed the syntax used in the message seems to be the intended way to specify the basic type.

I see now that I was feeding a Duration message to `max_capture_time` when it is expecting the 'duration' basic type. So presumably this would also be valid:

```
request = CaptureAssistantSuggestSettingsRequest(max_capture_time=Duration(2).data)
```

closed 2021-08-10 by danielcranston