zivid/zivid-ros #50: Ros 2 support

ranjitkathiriya, 2021-08-19:
Hey there,

Are you providing support for ros2?

runenordmo, 2021-08-20:
Hi, @ranjitkathiriya

Thanks for your interest in our ROS driver! Are you currently developing
a solution with any of the Zivid cameras? How will support for ROS 2 be
beneficial for you? Could an alternative like a network bridge 
(https://github.com/ros2/ros1_bridge) be suited for you?

Supporting ROS 2 is in our backlog, but we do not yet have any concrete plans 
for when to work on this. This is dependent on how many requests we get to 
support ROS 2.

Our main objective is to support LTS releases. Thus we would like to 
support ROS 2 at least in time for the Ubuntu 22.04 release.

We will post an update when supporting ROS 2 comes onto our roadmap.

ranjitkathiriya, 2021-08-23:
Hello @runenordmo,

> Are you currently developing a solution with any of the Zivid cameras?
No, We were planning, currently, we were using an intel real sense stereo camera. when we heard the news about shutting down real sense support so we are planning to switch to Zivid.

> How will support for ROS 2 be beneficial for you? 

Our full project is in ROS noetic at this moment but after few months we are planning to change it to foxy. 

> Could an alternative like a network bridge (https://github.com/ros2/ros1_bridge) be suited for you?

We cannot look at this because already we are using so much GPU for 3D pose identification so we want direct support. 

Anyways, for the current scenario intel-realsense is going to provide support to D400 series modules. I think it will be okay.

I am sorry for your time and thanks for your responce.

apockill, 2022-01-18:
Could this issue be re-opened? I'd also like to know the state of ros2 support, whether or not it's on it's way, and how I might be able to contribute.

runenordmo, 2022-01-28:
Hi, @apockill 

Thanks for your interest in our Zivid ROS wrapper! I will re-open this issue, since we aim to support ROS 2 with our ROS wrapper.

We have this task in our backlog, but we don't have any updates on the timeline yet, due to other Zivid SDK improvements higher on our road map.

One short-term alternative is to check out this unofficial fork of zivid-ros with some ROS 2 functionality: https://github.com/ra-mtp-ntnu/zivid-ros2. Note that Zivid has not written or tested that code, and we don't provide any support for it. But it can give you some inspiration to get a solution up and running until Zivid provides full ROS 2 support.

With regards to contributions, you can make a public fork of our zivid-ros repository. Then we'll be happy to take a look and give some feedback when we add support for ROS 2.

apockill, 2022-01-28:
Much appreciated, thank you for the update!

MatthijsBurgh, 2024-03-19:
ROS1 last release is close to 4 years old. I think it is time to start providing official ROS2 support.

apartridge, 2024-03-20:
Hi @MatthijsBurgh, we have ROS2 support for Zivid cameras high on the feature roadmap now. We don't have an exact date set right now that we are ready to communicate, however, I expect that we will communicate more details about ROS2 plans within some weeks.

fwarmuth, 2024-05-07:
Hey how much is "some" weeks?

apartridge, 2024-05-29:
Hi. Sorry for the late update. Good news, we are currently now working on ROS2 support for Zivid. We expect this to be ready for production usage around early July. We hope to have a early preview branch ready for testing and feedback within ca 2 weeks from now. 

We expect that preview to be fully functional, but we would like some time for feedback and testing ahead of the production release. We would be very interested in feedback from the users on that early preview, so that we can make any necessary adjustments to API's/behavior before the production ready driver is released and the API is locked for modifications.

For example, we are looking at simplifying how to set Zivid settings for 2D/3D captures, by only supporting path to .yml file and/or .yml string. In ROS1, we are supporting setting all of the Zivid setting nodes via dynamic_reconfigure as well, but we believe just supporting Yml files/strings would make it easier for users and easier for Zivid to maintain.

apartridge, 2024-06-17:
A branch with support for Zivid+ROS2 has been created here: https://github.com/zivid/zivid-ros/tree/ros2-preview. We hope that as many as possible are willing to test out this preview driver, and provide feedback/requests/bug reports to us.

We plan to release a official non-preview ROS2 driver in first half of July.

Stoiber-dkar, 2024-07-15:
Hi, any updates on the release?

fwarmuth, 2024-07-15:
you can use the ros2 branch, for me it workes flawlessly up to now.

fwarmuth, 2024-07-15:
you can use the ros2 branch, for me it worked flawlessly up to now.

apartridge, 2024-07-16:
Hi. We plan to update the master branch to ros2 (merging in https://github.com/zivid/zivid-ros/tree/ros2-preview) within a few weeks. We are also working on some improvements to the samples for ROS2. We do not expect any big changes in API or behavior compared to the current state of the branch. 

If anyone has any requests for changes to behavior/API in the ROS2 branch, let us know as soon as possible. After we have merged it to master we will keep the API stable.

apartridge, 2024-07-24:
ROS2 support is merged into master now.