zivid/zivid-ros #70: Capture with ROS Action Client and Server

waiyc, 2022-05-13:
Any plan to include ROS action client and server as an alternative option for capturing data from Zivid camera?
It is better option to use action over service ? As we can expect the action server to return the capturing result or we can choose to preempt it if we need to.

torbsorb, 2022-05-13:
Hi @waiyc,

As far as I understood the major difference between action and service is that actions are asynchronous, whereas services are synchronous. I can understand why it would be interesting for you, but
1. The APIs that our ROS-services map to are synchronous today
2. It is possible for a user to make the call asynchronous via threads, mutexes and signals.
Adding actions now might bring value, but would extend our API beyond what's provided in the core language. I believe it's more important to make sure that we have full coverage of our APIs, and work on potentially providing asynchronous functionality in the core language first. Then, once that's available it makes sense to support actions.

waiyc, 2022-05-17:
Hi @torbsorb ,

Thank you for the reply. My main intention is to increase the "certainty" in knowing when the capturing process is done or failed.
Currently, there is no response/feedback to tell the client that the server has completed the capturing. The only way to know the capturing process is completed is by subscribing to the image topic, is my understanding correct?

apartridge, 2022-06-24:
Hi @waiyc.

You could wait for the image or point cloud topics, but it would be better to check the return value of `ros::service::call()` function (if you use C++ API) or `rospy.ServiceProxy` for the Python API. These functions should be blocking until the capture service is finished and has returned either true or false.