zivid/zivid-ros #74: ros noetic catkin build error

taisi5446, 2022-08-30:
I tried to build zivid-ros in noetic.

I was installed zivid core SDK with 4 deb files and all dependency of zivid-ros for rosdep. But catkin build faced some error like below.

How solve it?

erp42@erp42:~/catkin_ws$ catkin build zivid*
----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/erp42/catkin_ws
----------------------------------------------------------
Build Space:        [exists] /home/erp42/catkin_ws/build
Devel Space:        [exists] /home/erp42/catkin_ws/devel
Install Space:      [unused] /home/erp42/catkin_ws/install
Log Space:          [exists] /home/erp42/catkin_ws/logs
Source Space:       [exists] /home/erp42/catkin_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------
[build] Found 25 packages in 0.0 seconds.                                      
[build] Package table is up to date.                                           
Starting  >>> zivid_camera                                                     
_______________________________________________________________________________
Errors     << zivid_camera:make /home/erp42/catkin_ws/logs/zivid_camera/build.make.003.log
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/geneus/geneus_main.py", line 136, in genmain
    pkg_map = get_pkg_map()
  File "/opt/ros/noetic/lib/python3/dist-packages/geneus/geneus_main.py", line 56, in get_pkg_map
    pkgs = packages.find_packages(ws)
  File "/usr/lib/python3/dist-packages/catkin_pkg/packages.py", line 103, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "actionlib":
- include/opt/ros/noetic/share/actionlib
- share/actionlib
Multiple packages found with the same name "actionlib_msgs":
- include/opt/ros/noetic/share/actionlib_msgs
- share/actionlib_msgs
Multiple packages found with the same name "actionlib_tutorials":
- include/opt/ros/noetic/share/actionlib_tutorials
- share/actionlib_tutorials
Multiple packages found with the same name "angles":
- include/opt/ros/noetic/share/angles
- share/angles
Multiple packages found with the same name "apriltag_ros":
- include/opt/ros/noetic/share/apriltag_ros
- share/apriltag_ros
Multiple packages found with the same name "bond":
- include/opt/ros/noetic/share/bond
- share/bond
Multiple packages found with the same name "bond_core":
- include/opt/ros/noetic/share/bond_core
- share/bond_core
Multiple packages found with the same name "bondcpp":
- include/opt/ros/noetic/share/bondcpp
- share/bondcpp
Multiple packages found with the same name "bondpy":
[... 719 lines cut ...]
Multiple packages found with the same name "webkit_dependency":
- include/opt/ros/noetic/share/webkit_dependency
- share/webkit_dependency
Multiple packages found with the same name "xacro":
- include/opt/ros/noetic/share/xacro
- share/xacro
Multiple packages found with the same name "xmlrpcpp":
- include/opt/ros/noetic/share/xmlrpcpp
- share/xmlrpcpp
make[2]: *** [CMakeFiles/zivid_camera_generate_messages_eus.dir/build.make:81: /home/erp42/catkin_ws/devel/.private/zivid_camera/share/roseus/ros/zivid_camera/manifest.l] Error 3
make[1]: *** [CMakeFiles/Makefile2:1796: CMakeFiles/zivid_camera_generate_messages_eus.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
cd /home/erp42/catkin_ws/build/zivid_camera; catkin build --get-env zivid_camera | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << zivid_camera:make            [ Exited with code 2 ]              
Failed    <<< zivid_camera                 [ 0.6 seconds ]                     
Abandoned <<< zivid_samples                [ Unrelated job failed ]            
[build] Summary: 0 of 2 packages succeeded.                                    
[build]   Ignored:   23 packages were skipped or are skiplisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 0.7 seconds total.

apartridge, 2022-09-03:
Hi. You have many variants of this error for many packages:
```
RuntimeError: Multiple packages found with the same name "actionlib":

include/opt/ros/noetic/share/actionlib
share/actionlib
```
So it appears it finds multiple packages with the same names in your workspace.
I haven't seen your error before, but I see several results when googling for "Multiple packages found with the same name".
Can you check if any of these are applicable for you?

We do build our driver with ros-noetic in our CI and do works (but sometimes tests fails for different reason). So it should work to build our driver from a clean machine & workspace.

apartridge, 2024-08-05:
Closing this issue due to no response from reporter, and since we have now moved to ROS2.

closed 2024-08-05 by apartridge