zivid/zivid-ros #84: Zivid nodelet can't attach to a manager in the global namespace

danielcranston, 2023-04-05:
### Summary
Hi,

I think it's great that the driver itself enforces [REP-135](https://www.ros.org/reps/rep-0135.html), but I noticed a situation where this enforcement is too strict.

It has to do with the nodelet version of the driver. The method the driver uses to determine if it was started in the global namespace (the thing that REP-135 prohibits) is `ros::this_node::getNamespace()`.

For nodelets, this resolves to the namespace of the nodelet _manager_, not the _Zivid driver nodelet_ (at least in ROS Melodic). This means that a Zivid driver nodelet, brought up in its own namespace, will fail to start (claiming it doesn't satisfy REP-135, but in fact it does).

I made a commit with a proposed solution here: https://github.com/zivid/zivid-ros/commit/473bf4eb9ac7727c58f4302349b87b673e21bff9

### How to reproduce

To test, you can use this minimal launchfile

```
<launch>
    <arg name="ns" default="/" />

    <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager" output="screen" />
    <node ns="$(arg ns)" pkg="nodelet" type="nodelet" name="latch" args="load zivid_camera/nodelet /nodelet_manager" output="screen" />
</launch>
```

Notice how the manager is in the global namespace and the Zivid driver nodelet is put in `$(arg ns)`.

`roslaunch test.launch ns:=zivid` fails on `master`, but succeeds on https://github.com/zivid/zivid-ros/commit/473bf4eb9ac7727c58f4302349b87b673e21bff9. Starting the nodelet (or node) in the global namespace still has the same (intended) behavior as before. 

I guess one could make the argument that nodelet managers shouldn't be put in the global namespace, but I don't interpret anything in REP-135 that would discourage it, and I think it's a pretty common practice to do so.

Thanks for reading.

apartridge, 2024-06-12:
Hi @danielcranston , thanks for your report and fix. I have added a PR for it here https://github.com/zivid/zivid-ros/pull/98 and we will aim to merge it in soon.

apartridge, 2024-06-14:
Solved with https://github.com/zivid/zivid-ros/pull/99

closed 2024-06-14 by apartridge