zivid/zivid-ros #104: How to mount the calibration board to UR5e

hai-h-nguyen, 2024-07-03:
Hi,

Thanks for an excellent camera, I really enjoy the quality of the images. I am trying to do the calibration as described in https://support.zivid.com/en/latest/academy/applications/hand-eye/ur5-robot-%2B-python-generate-dataset-and-perform-hand-eye-calibration.html. However, in the tutorial, it is unclear how the calibration board is mounted onto the robot? Do I need a specific mount for that purpose? Also, it seems that there are four holes on the board which can be directly mounted on the last link of the robot but I am not sure if it is the way.

Thank you!

hai-h-nguyen, 2024-07-03:
It seems that there is a special mounting mechanism here, https://support.zivid.com/en/latest/academy/applications/hand-eye/hand-eye-calibration-process.html. Did you guys sell it, or if not, is there a model which I can 3D-print? Thanks!

torbsorb, 2024-07-03:
As you correctly figured out there are mounting holes on the board. The hand-eye algorithm does not depend on a specific way of mounting, as long as the board is visible to the camera. I would recommend mounting it so that it faces the camera while the flange is pointing towards where it will be during operation. Some customers design special mounts, but that's often for convenience.

When you perform hand-eye you want to use poses that are as close to those you will use during operation, and that covers the whole working area you plan to operate in.

hai-h-nguyen, 2024-07-03:
Thanks for your quick reply! I also just figured out that the mounting does not matter as long as it does not change among poses. Thank you again.

closed 2024-07-03 by hai-h-nguyen

hai-h-nguyen, 2024-07-05:
Hi, I just performed the calibration using the tutorial and the result looks like this:

  Data: [
    [-0.003973591, 0.9993435, 0.03601083, 25.07366],
    [0.9989207, 0.005633254, -0.04610426, -719.6547],
    [-0.04627685, 0.03578877, -0.9982873, 1197.474],
    [0, 0, 0, 1]]

Does it seem wrong? Should the diagonal values close to 1?