zivid/zivid-ros #105: Using std_srvs instead of custom ones

belalhmedan90, 2024-07-04:
Hi, 
I want to use your ROS wrapper to avoid using the SDK directly and use ROS instead, 
since your camera type is triggered one and not continous stream, I have to trigger two services:
1. To set/find/load the configurations.
2. To capture the image

However you are defining custom services, hence I should add dependency to your ROS driver!

1. Can you please change the Empty services (e.g [Capture](https://github.com/zivid/zivid-ros/blob/master/zivid_camera/srv/Capture.srv), [Capture2D](https://github.com/zivid/zivid-ros/blob/master/zivid_camera/srv/Capture2D.srv) ) to the standard [std_srvs/Empty](http://docs.ros.org/en/noetic/api/std_srvs/html/srv/Empty.html)
2. Is there any possibility to provide a standard service similar to [CaptureAssistantSuggestSettings]([CaptureAssistantSuggestSettings.srv](https://github.com/zivid/zivid-ros/blob/master/zivid_camera/srv/CaptureAssistantSuggestSettings.srv)) with default parameters so we avoid using custom services? thanks.

apartridge, 2024-07-04:
HI @belalhmedan90 . Are you using ROS1 or ROS2? In our ROS2 preview we are using std_srvs for the services that don't have any parameters, see https://github.com/zivid/zivid-ros/tree/ros2-preview.

belalhmedan90, 2024-07-05:
> HI @belalhmedan90 . Are you using ROS1 or ROS2? In our ROS2 preview we are using std_srvs for the services that don't have any parameters, see https://github.com/zivid/zivid-ros/tree/ros2-preview.

Thanks for your rapid reply, Actually it is for ROS1

belalhmedan90, 2024-07-05:
@apartridge Moreover I saw the ROS2 preview with cool interfaces package, and I suggest that the service [IsConnected](https://github.com/zivid/zivid-ros/blob/ros2-preview/zivid_interfaces/srv/IsConnected.srv) becomes a standard [Trigger](http://docs.ros.org/en/noetic/api/std_srvs/html/srv/Trigger.html) service if possible

Also I didn't find the service to load the 2D, 3D YML configurations from a file.

apartridge, 2024-07-05:
Hi @belalhmedan90. in the ROS2 driver preview there are parameter values you can set to the .yml files for 2D and 3D captures. Since we support .yml files via parameters, we decided to not have a separate service for loading a .yml file.

As for IsConnected, to me it sounds like using `Trigger` is a bit misleading since that service is not triggering anything, just checking the current state of the driver? Also be aware that in the ROS2 driver we moved the srv files to a separate package, which makes it easier to depend on that package and not the entire zivid_camera package.

closed 2024-07-05 by belalhmedan90