zivid/zivid-ros #108: Rotation Order in Calibration Matrix

hai-h-nguyen, 2024-07-08:
Hi,

I performed hand-to-eye calibration and got this result:

Data: [
[-0.003973591, 0.9993435, 0.03601083, 25.07366],
[0.9989207, 0.005633254, -0.04610426, -719.6547],
[-0.04627685, 0.03578877, -0.9982873, 1197.474],
[0, 0, 0, 1]]

From the documentation, I understand this is the pose of the camera in the robot's base coordinate. In my case, the translation vector seems wrong as translation_y and translation_z should both be positive as the camera is mounted on top and in the same direction with the y-axis of the robot base. However, I used the verification script to check and the Can you explain this? Also, I want to know what is the order of the rotations in the matrix. Thanks!
[image]

SatjaSivcev, 2024-07-09:
What Zivid code sample have you used for hand-eye and for verification?

P.S. We can likely resolve your issue faster if you contact us at customersuccess@zivid.com for support as we can then also organize an online call to help you out.

hai-h-nguyen, 2024-07-09:
Hi thanks. Do I need any information of the product  for that assisstance? I used the robodk tutorial for calibration https://support.zivid.com/en/latest/academy/applications/hand-eye/robodk-%2B-python-generate-dataset-and-perform-hand-eye-calibration.html and verification using the visualization inside that tutorial.

SatjaSivcev, 2024-07-09:
Based on your screenshot (of the UR robot), the robot base coordinate system is such that x (red) goes in front of the robot, y(green) to the left, and z (blue, not seen) goes upwards. Based on this and based on your hand-eye transform, the camera should be above the robot (positive z), a bit in front of the robot (positive x) and somewhat right of the robot (negative y). Is this not how you have it in the real world? This is not where "zivid_optical_frame" seems to be based on the screenshot.

You are correct, our eye-to-hand outputs pose of camera in the robot base frame.

"Do I need any information of the product for that assistance?" Not necessarily, but we would normally like to know your affiliation and country so we can assign the right person for the support case.

hai-h-nguyen, 2024-07-09:
Hi, thanks. I'll send an email. The figure is the correct configuration in the real-world in terms of the two coordinates (I left orientations to be all 0s). The figure is however when I change the y-offset to be positive (the calibration result had it negative). Why did you say y-offset should be negative? I think it should be positive right, because the coordinate is on the positive side of the robot's base y direction.

SatjaSivcev, 2024-07-09:
> Why did you say y-offset should be negative?

We will continue via email then. Just to reply:
- "Why did you say y-offset should be negative?" Because that is what I see in the hand-eye output (-719.6547)
- "when I change the y-offset to be positive (the calibration result had it negative)" changing the sign of one axis doesn't seem like a good idea.

I am wondering where you get robot poses from (UR pendant, or ROS interface, etc.), but this is best discussed in the email thread and potentially an online meeting.

hai-h-nguyen, 2024-07-09:
Hi, thanks. I sent the email asking for help. For your questions, I had to change the sign to match the real-world configuration. Also, I got the pose from the UR pendant (using RoboDK tutorial https://support.zivid.com/en/latest/academy/applications/hand-eye/robodk-%2B-python-generate-dataset-and-perform-hand-eye-calibration.html actually, not manually). I look forward to the help via the other channel.

hai-h-nguyen, 2024-07-09:
Hi, figured it out. For future reference, the result of the calibration is the transform from zivid_optical_frame to the robot_base_inertia coordinate in ROS (not robot_base). Everything works now.

closed 2024-07-09 by hai-h-nguyen