zivid/zivid-ros #113: ROS2 - zivid_camera node dies right after launching it

dk-teknologisk-gtr, 2024-08-22:
Hi, I am experiencing some issues when running the camera driver.
I am running the branch master in ROS2 Humble.
I have cloned the repo and build it in my workspace. The SDK is correctly installed in my Ubuntu 22 machine.
I can see the camera and take pointcloud data correctly with ZividStudio.

```bash:
$ ros2 run zivid_camera zivid_camera
[ros2run]: Trace/breakpoint trap
```

Also when I try to use one of the launch file provided:
```bash
$ ros2 launch zivid_samples sample_with_rviz.launch sample:=sample_capture_cpp

[INFO] [launch]: All log files can be found below /root/.ros/log/2024-08-22-12-34-57-568827-ubuntu20-149843
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [zivid_camera-1]: process started with pid [149847]
[INFO] [sample_capture_cpp-2]: process started with pid [149849]
[INFO] [rviz2-3]: process started with pid [149851]
[sample_capture_cpp-2] [INFO] [1724330097.690943148] [sample_capture_cpp]: Started the sample_capture node
[sample_capture_cpp-2] [INFO] [1724330097.691008280] [sample_capture_cpp]: Setting parameter 'settings_yaml'
[rviz2-3] [INFO] [1724330097.818892038] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1724330097.818953988] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-3] [INFO] [1724330097.832721753] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1724330097.935993822] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1724330097.950747737] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [zivid_camera-1]: process has died [pid 149847, exit code -5, cmd '/ros2_ws/install/zivid_camera/lib/zivid_camera/zivid_camera --ros-args -r __node:=zivid_camera'].
```

Is there anything I am doing wrong?

To add a bit more on my question this is the output of running the `zivid_camera` driver with `gdb`:

```bash:
$ ros2 run --prefix 'gdb -ex run --args' zivid_camera zivid_camera
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /ros2_ws/install/zivid_camera/lib/zivid_camera/zivid_camera...
(No debugging symbols found in /ros2_ws/install/zivid_camera/lib/zivid_camera/zivid_camera)
Starting program: /ros2_ws/install/zivid_camera/lib/zivid_camera/zivid_camera 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6873640 (LWP 154009)]
[New Thread 0x7fffe78c1640 (LWP 154010)]
[New Thread 0x7fffe7039640 (LWP 154011)]
[New Thread 0x7fffe6838640 (LWP 154012)]
[New Thread 0x7fffe6037640 (LWP 154013)]
[New Thread 0x7fffe5836640 (LWP 154014)]
[New Thread 0x7fffe5035640 (LWP 154015)]
[New Thread 0x7fffe4774640 (LWP 154016)]
[New Thread 0x7fffe3f5e640 (LWP 154031)]
[New Thread 0x7fffe3330640 (LWP 154036)]
[New Thread 0x7fffe2b2f640 (LWP 154037)]
[New Thread 0x7fffe232e640 (LWP 154038)]
[Thread 0x7fffe2b2f640 (LWP 154037) exited]
[New Thread 0x7fffbb463640 (LWP 154039)]
[New Thread 0x7fffb9c41640 (LWP 154040)]
[New Thread 0x7fffb9240640 (LWP 154041)]
[New Thread 0x7fffb8a3f640 (LWP 154042)]
[New Thread 0x7fffb1fff640 (LWP 154043)]
[New Thread 0x7fffb17fe640 (LWP 154044)]
[New Thread 0x7fffb0ffd640 (LWP 154045)]

Thread 1 "zivid_camera" received signal SIGSEGV, Segmentation fault.
0x00007ffff601e788 in zivid_camera::ZividCamera::setParametersCallback(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) () from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
(gdb) bt
#0  0x00007ffff601e788 in zivid_camera::ZividCamera::setParametersCallback(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) ()
   from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
#1  0x00007ffff602cc3c in std::_Function_handler<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&), std::_Bind<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (zivid_camera::ZividCamera::*(zivid_camera::ZividCamera*, std::_Placeholder<1>))(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&)> >::_M_invoke(std::_Any_data const&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) () from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
#2  0x00007ffff7ed0402 in ?? () from /opt/ros/humble/lib/librclcpp.so
#3  0x00007ffff7ed2c63 in ?? () from /opt/ros/humble/lib/librclcpp.so
#4  0x00007ffff7ed354d in ?? () from /opt/ros/humble/lib/librclcpp.so
#5  0x00007ffff7ed3cb7 in rclcpp::node_interfaces::NodeParameters::declare_parameter(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::ParameterValue const&, rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> > const&, bool) () from /opt/ros/humble/lib/librclcpp.so
#6  0x00007ffff6038c2a in auto rclcpp::Node::declare_parameter<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> > const&, bool) ()
   from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
#7  0x00007ffff6041caf in zivid_camera::CaptureSettingsController<Zivid::Settings>::CaptureSettingsController(rclcpp::Node&) () from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
#8  0x00007ffff6026892 in zivid_camera::ZividCamera::ZividCamera(rclcpp::NodeOptions const&) () from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
#9  0x00007ffff605083c in rclcpp_components::NodeFactoryTemplate<zivid_camera::ZividCamera>::create_node_instance(rclcpp::NodeOptions const&) ()
   from /ros2_ws/install/zivid_camera/lib/libzivid_camera_component.so
#10 0x0000555555558215 in main ()
(gdb) continue
Continuing.

Thread 1 "zivid_camera" received signal SIGTRAP, Trace/breakpoint trap.
__pthread_kill_implementation (no_tid=0, signo=5, threadid=140737344065344) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) continue
Continuing.
[Thread 0x7fffe6838640 (LWP 154012) exited]
[Thread 0x7fffe6037640 (LWP 154013) exited]
[Thread 0x7fffe5836640 (LWP 154014) exited]
[Thread 0x7fffe5035640 (LWP 154015) exited]
[Thread 0x7fffe7039640 (LWP 154011) exited]
[Thread 0x7fffe4774640 (LWP 154016) exited]
[Thread 0x7fffe78c1640 (LWP 154010) exited]
[Thread 0x7fffb0ffd640 (LWP 154045) exited]
[Thread 0x7fffb17fe640 (LWP 154044) exited]
[Thread 0x7fffb1fff640 (LWP 154043) exited]
[Thread 0x7fffb8a3f640 (LWP 154042) exited]
[Thread 0x7fffb9240640 (LWP 154041) exited]
[Thread 0x7fffb9c41640 (LWP 154040) exited]
[Thread 0x7fffbb463640 (LWP 154039) exited]
[Thread 0x7fffe232e640 (LWP 154038) exited]
[Thread 0x7fffe3f5e640 (LWP 154031) exited]
[Thread 0x7ffff6873640 (LWP 154009) exited]
[Thread 0x7ffff7664f40 (LWP 154006) exited]
[Thread 0x7fffe3330640 (LWP 154036) exited]
[New process 154006]
[Inferior 1 (process 154006) exited with code 01]
(gdb) 

```

By doing a bit of debugging it I see that I have to provide a setting yaml file from ZividStudio, how can I provide that directly in the launch file? I don't see any example in the package `zivid_samples`.

Thank you in advance for your help!

apartridge, 2024-08-22:
Hi @dk-teknologisk-gtr, could you zip all the files you find in  ~/.cache/Zivid/API/Log and send them to me at stian.pedersen (at) zivid.com? It may contain some extra information. I will try to reproduce with the same commands on my PC, I am out of office tomorrow, but I will test this early next week.

> By doing a bit of debugging it I see that I have to provide a setting yaml file from ZividStudio, how can I provide that directly in the launch file? I don't see any example in the package zivid_samples.

With our ROS2 driver you can specify settings either as a path to a YAML File or a YAML multi-line string (basically copy the contents of a settings.yml file from Zivid Studio). Use 'settings_yaml' when you want to use a YAML string, and 'settings_file_path' when you want to use a .yml file. We will look at making a launch file example for this.

dk-teknologisk-gtr, 2024-08-22:
Yes, I am actually running it in a docker container with Ubuntu 22.04 and Ros2 Humble.
I have followed the step in https://support.zivid.com/en/latest/api-reference/docker/install-zivid-in-docker.html (NVIDIA settings) one month ago (with the ROS1 version of the driver) and everything worked just fine.

I have the SDK installed also in my host machine (Ubuntu 20.04).

johningve, 2024-08-22:
Looks like a memory bug:

```
#0  0x00007a44a931a928 in std::_Optional_base_impl<Zivid::Settings2D, std::_Optional_base<Zivid::Settings2D, false, false> >::_M_is_engaged (this=0x68) at /usr/include/c++/11/optional:433
#1  0x00007a44a931a3c6 in std::optional<Zivid::Settings2D>::has_value (this=0x68) at /usr/include/c++/11/optional:945
#2  0x00007a44a9310dfc in zivid_camera::CaptureSettingsController<Zivid::Settings2D>::onSetParameter (this=0x20, parameterName="settings_yaml") at /root/ros2_ws/src/zivid-ros/zivid_camera/src/zivid_camera.cpp:274
#3  0x00007a44a92fc1ec in zivid_camera::ZividCamera::setParametersCallback (this=0x5c9fc99752b0, parameters=std::vector of length 1, capacity 1 = {...}) at /root/ros2_ws/src/zivid-ros/zivid_camera/src/zivid_camera.cpp:530
530           settings_2d_controller_->onSetParameter(param.get_name());
(gdb) p settings_2d_controller_
$3 = std::unique_ptr<zivid_camera::CaptureSettingsController<Zivid::Settings2D>> = {get() = 0x20}
(gdb)
```

johningve, 2024-08-22:
Can you try the fix in https://github.com/zivid/zivid-ros/pull/114 and see if that resolves the issue?

dk-teknologisk-gtr, 2024-08-23:
Hi @johningve, yes that fixed the problem. Now I can connect to the camera and correctly read parameters file!
Thank you

falfab, 2024-08-29:
Hi @johningve, had the same issue using ROS Humble on Ubuntu 22.04.

#114 solved the issue for me. 

Thanks

apartridge, 2024-09-09:
https://github.com/zivid/zivid-ros/pull/114 is now merged, so this issue should be fixed.

closed 2024-09-09 by apartridge