zivid/zivid-ros #116: Camera node exits on failed service requests

dave992, 2024-10-01:
The `zivid_camera` node exits when it is unable to perform a service request. 

### Minimal working example:

Terminal 1:
```bash
ros2 run zivid_camera zivid_camera
```

Terminal 2:
```bash
ros2 service call /capture std_srvs/srv/Trigger "{}"
```

The node in terminal 1 displays an error and exits. The error is the result of not having any frames configured for the capture, this is expected. The node exits before finishing the service call, resutling in terminal 2 hanging. 

### Expected behavior:
The service returns with a failure indication and does not exit. For example, the service response could be: 
```
succes=False
message="No frames are configured, cannot take a capture. Configure frames first"
```
This is supported by the `std_srvs/Trigger` message type that is used already. 

Additional service calls should be possibe after this again.

apartridge, 2024-10-02:
Hi @dave992, thanks for your bug report.

When I test the same here, using ROS2 Humble on Ubuntu 22.04, capture initailly fails because no settings are defined:
```
[ERROR] [1727874791.762142499] [zivid_camera]: Both 'settings_file_path' and 'settings_yaml' parameters are empty! Please set one of these parameters.
[ERROR] [1727874791.762188929] [zivid_camera]: A function failed with exception: "Both 'settings_file_path' and 'settings_yaml' parameters are empty! Please set one of these parameters."
```

When I set the settings using a .yml file (exported from Studio) when starting the driver;

```
ros2 run zivid_camera zivid_camera --ros-args -p settings_file_path:=/home/stian/ros2_ws/settings.yml
```

then capture returns successfully. 
```
response:
std_srvs.srv.Trigger_Response(success=True, message='')
```

I am a bit unsure how to set it up to get that "No frames are configured" error. I am not so familiar with the tf2 framework, it may be that a specific state is needed to get this issue?

dave992, 2024-10-02:
In general, for any service that fails (for any reason) in a non-fatal way, I would expect the following to be true:
- :white_check_mark: Requested service should not complete
- :white_check_mark: Node should display/log an error
- :x: Node should return a result
- :x: Node should stay running and services / topics / etc. should stay available

Furthermore, I think that the result should always indicate wether it was successful and if not, why. This is actually possible with the `std_srv/Trigger` service due to the `success` and `message` fields.

dave992, 2024-10-02:
Ah okay. Yes, it was indeed the missing `settings.yml`. I think it was also exiting when I provided an incorrect version of the settings file, but after your explanation I am no longer sure.

I've also had the node exit when invoking the `capture_assistant/suggest_settings` service from the terminal in an attempt to quickly fix the issue. This failed due to an incorrect value for `ambient_light_frequency`. This made me believe this was more general behavior instead of just a specific case. Thanks for the clarification.

Would it be an option to not have the exception propagate then? It is still recoverable by setting one of the ROS parameters or by calling the `capture_assistant/suggest_settings`. I also do not have a way to check if my settings are valid and accepted until I call the service as far as I know.

apartridge, 2024-10-14:
Hi @dave992, agree that it is a bit too "strict" and not so user friendly to kill the ROS node if the settings parameters are not set. I will take a look at this soon.

dave992, 2024-10-14:
Thanks, that would be great 👍

apartridge, 2024-10-15:
Merged in https://github.com/zivid/zivid-ros/pull/117

closed 2024-10-15 by apartridge