zivid/zivid-ros #118: Difference in Image Contrast: Zivid Studio vs. ROS2 /capture Service

kumar-sanjeeev, 2024-11-20:
Description:

Hello,

We are using the Zivid 2+ M60 model and have observed a discrepancy while working with the camera. Specifically, there is a noticeable difference in image contrast between images captured using `Zivid` Studio and those captured via the ROS2  `/capture` service , despite using same camera settings. We would like to confirm whether this is expected behavior or if we might be missing something.

Scenario 1:
- Method: Image captured using Zivid Studio.
[image]

- Observation: (Sample image attached).

Scenario 2:
- Method: Image captured using the ROS2 /capture service.
- Settings: Identical camera settings as used in Zivid Studio.
- Observation: Image contrast differs from Scenario 1 (Rosbag file attached).

 [rosbag file link](https://studentsrwthaachende-my.sharepoint.com/:u:/g/personal/mrurn7kd8v6pxt5r_students_rwth-aachen_de/ETbtG4SLdkhGqu2VXiK_KbkBjlnKMkCVBoX3JsKpONgsng?e=xplANi)

apartridge, 2024-11-20:
Hi @kumar-sanjeeev,  thanks for your ticket. 

The color difference comes from Studio showing the colors using sRGB color space, which makes it appear brighter (and more correct on the monitor). In the ROS driver the colors outputted have not been converted to sRGB. You  can read more about it at https://support.zivid.com/en/latest/support/color-inconsistency-after-sdk-update.html#for-all-cameras-after-updating-to-sdk-2-11 and https://support.zivid.com/en/latest/reference-articles/color-spaces-and-output-formats.html.

We will look at add an option (topic) to the ROS driver to retrieve the colors in sRGB format. Those colors would then look identical as in Studio (for the same settings). You can then use that topic to get colors that match Studio. However, for your color processing you need to evaluate if you want to use colors in linear format  or in SRGB format, depending on what your color processing pipeline is optimized for.

A workaround for now is to set the Gamma setting in Studio to 0.45 which approximately equals sRGB. ROS colors would then look similar to the Studio colors when Gamma was 1.0 (Studio colors would then look very bright with Gamma 0.45). 

A fix in the code would be to replace Zivid::ColorRGBA with Zivid::ColorSRGB in https://github.com/zivid/zivid-ros/blob/master/zivid_camera/src/zivid_camera.cpp and https://github.com/zivid/zivid-ros/blob/master/zivid_camera/include/zivid_camera/zivid_camera.hpp. This would change the output for the `color/image_color` topic to use sRGB instead of linear RGB, but it would not affect the `points/xyzrgba` topic.

kumar-sanjeeev, 2024-11-25:
Hello @apartridge, thanks for suggesting a way to solve problem. I tried and it worked :)

closed 2024-11-25 by kumar-sanjeeev

EmmaHeyen, 2025-03-11:
Hello @apartridge , i tried this too, but it did not seem to work. More specifically, I replaced Zivid::ColorRGBA with Zivid::ColorSRGB in lines 821 and 828 in zivid_camera.cpp and in line 132 in zivid_camera.hpp

Could you tell me where i went wrong?

I get the following error when (colcon) building: 

--- stderr: zivid_camera                             
/home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/src/zivid_camera.cpp: In lambda function:
/home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/src/zivid_camera.cpp:596:26: error: no matching function for call to ‘zivid_camera::ZividCamera::publishColorImage(const std_msgs::msg::Header_<std::allocator<void> >&, const std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> > >&, Zivid::Image<Zivid::ColorRGBA>&)’
  596 |         publishColorImage(header, camera_info, image);
      |         ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/src/zivid_camera.cpp:46:
/home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/include/zivid_camera/zivid_camera.hpp:125:8: note: candidate: ‘void zivid_camera::ZividCamera::publishColorImage(const Header&, const ConstSharedPtr&, const Zivid::PointCloud&)’
  125 |   void publishColorImage(
      |        ^~~~~~~~~~~~~~~~~
/home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/include/zivid_camera/zivid_camera.hpp:128:31: note:   no known conversion for argument 3 from ‘Zivid::Image<Zivid::ColorRGBA>’ to ‘const Zivid::PointCloud&’
  128 |     const Zivid::PointCloud & point_cloud);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/include/zivid_camera/zivid_camera.hpp:129:8: note: candidate: ‘void zivid_camera::ZividCamera::publishColorImage(const Header&, const ConstSharedPtr&, const Zivid::Image<Zivid::ColorSRGB>&)’
  129 |   void publishColorImage(
      |        ^~~~~~~~~~~~~~~~~
/home/user/ros2_ws_zividcam/src/zivid-ros/zivid_camera/include/zivid_camera/zivid_camera.hpp:132:44: note:   no known conversion for argument 3 from ‘Zivid::Image<Zivid::ColorRGBA>’ to ‘const Zivid::Image<Zivid::ColorSRGB>&’
  132 |     const Zivid::Image<Zivid::ColorSRGB> & image);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
gmake[2]: *** [CMakeFiles/zivid_camera_component.dir/build.make:76: CMakeFiles/zivid_camera_component.dir/src/zivid_camera.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:159: CMakeFiles/zivid_camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zivid_camera [3.07s, exited with code 2]

micragz, 2025-03-11:
Hi @EmmaHeyen. Can you try to replace the following line:

https://github.com/zivid/zivid-ros/blob/330d34100fc9ee1f2033731b2a189a3133af343b/zivid_camera/src/zivid_camera.cpp#L593

with: 

```cpp
auto image = frame2D.imageSRGB();
```

and see if it builds now?

We aim to provide a better solution in the future.

micragz, 2025-04-07:
Hello,

We have now merged #131 which should help with this issue. It is now possible to set the `color_space` parameter on the zivid camera node to `srgb`. The published colors will then be provided in the sRGB color space, thereby matching what can be seen in Zivid Studio.