zivid/zivid-ros #123: The time taken to connect to the camera is too long

wilsonxiao, 2025-01-02:
When I run zivid_camera, I find that it takes more than 1 minute to connect to the camera. 
Here are the log details:
wilson@MyComputer:~/ros2_ws/src$ ros2 run zivid_camera zivid_camera 
[INFO] [1735653256.641514510] [zivid_camera]: Zivid ROS driver
[INFO] [1735653256.641668080] [zivid_camera]: The node's namespace is '/'
[INFO] [1735653256.641680777] [zivid_camera]: Running Zivid Core version 2.13.1+18e79e79-1
[INFO] [1735653259.650787966] [zivid_camera]: 1 camera(s) found
[INFO] [1735653259.651053495] [zivid_camera]: Selecting first available camera
[INFO] [1735653259.652336520] [zivid_camera]: { Serial Number: 234470BG, Status: available, Model: zividTwo, Model Name: Zivid 2 M70, IP Address: 192.168.0.171, Firmware Version: 1.26.3+ff1e16c8, Hardware Revision: B3 }
[INFO] [1735653259.652709964] [zivid_camera]: Connecting to camera '234470BG'
[INFO] [1735653331.557409080] [zivid_camera]: Connected to camera '234470BG'
[INFO] [1735653331.557483683] [zivid_camera]: Camera status changed to Connected (was Idle)
[INFO] [1735653331.557676806] [zivid_camera]: Advertising topics
[INFO] [1735653331.561061099] [zivid_camera]: Advertising services
[INFO] [1735653331.564193092] [zivid_camera]: Zivid camera driver is now ready!

micragz, 2025-01-02:
Hi @wilsonxiao. The connection time can be quite long the first time the camera connect, particularly due to OpenCL kernels being compiled during this process. When connecting again from the same process it should be much faster. Some drivers also cache these kernels between runs. Please take a look at the following knowledge base article for more details, and instructions for enabling the mentioned cache: https://support.zivid.com/en/latest/support/long-connection-time.html

wilsonxiao, 2025-01-06:
Hello @micragz , thanks for the solution you provided. I tried and it worked.

closed 2025-01-06 by wilsonxiao