zivid/zivid-ros #127: Difference in Image Brightness and Effects Between Code Capture and ZividStudio

wilsonxiao, 2025-02-14:
Why do the images captured through code have different brightness and other visual effects compared to the ones captured using ZividStudio, even though the camera settings are the same in both cases?

The code-captured image：

[image]

The ZividStudio-captured image:

[image]

johningve, 2025-02-14:
Hi @wilsonxiao

It is due to Studio using sRGB color mode by default. You can see more information here: https://github.com/zivid/zivid-ros/issues/118#issuecomment-2488932246

micragz, 2025-04-07:
Hi @wilsonxiao,

We have now merged #131 which should help with this issue. It is now possible to set the `color_space` parameter on the zivid camera node to `srgb`. The published colors will then be provided in the sRGB color space, thereby matching what can be seen in Zivid Studio.

apartridge, 2025-04-10:
Closing this issue

closed 2025-04-10 by apartridge